DroneEngineerTeaParty/Drone_technology_usage #21: GCS: ESP8266でテレメトリする方法

muramura, 2019-01-05:
ESP8266互換WIFIモジュールでテレメトリする方法が書きURLに掲載されています。
https://docs.px4.io/en/telemetry/esp8266_wifi_module.html
＝＝＝
例：
murat@MSI:/mnt/d/aaaa$ sudo esptool.py --baud 921600 --port /dev/ttyS5 write_flash 0x00000 firmware-1.1.1.bin
esptool.py v2.5.1
Serial port /dev/ttyS5
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: XX:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 277440 bytes to 198576...
Wrote 277440 bytes (198576 compressed) at 0x00000000 in 2.5 seconds (effective 873.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
murat@MSI:/mnt/d/aaaa$

muramura, 2019-01-10:
@cubicrootk さんへ
教えてください。同じファームウェアを書き込むことができたということでしょうか。。

cubicrootk, 2019-01-11:
WROOM-02 は ESP8266で TELEC認証が入っているバージョンなので、APのWiKi通りで動作します。

muramura, 2019-01-17:
http://mag.switch-science.com/2016/01/20/esp-wroom-02_telec/
ファームウェアを書き換える場合、このルールにあっていれば、とてもいいですね。